riscvarchive/riscv-linux #17: "uname -m" reports "riscv", not "riscv64"

palmer-dabbelt, 2015-01-30:
This causes config.{guess,sub} to fail because the upstreamed tuple is "riscv64-unknown-linux-gnu", not "riscv-unknown-linux-gnu".  There's a super easy fix: apply the following patch and "make ARCH=riscv64", but since the patch touches "linux/Makefile" it can't be merged here because of how riscv-linux.git is setup.

````
diff --git a/Makefile b/Makefile
index 95a0e82..6e2ff96 100644
--- a/Makefile
+++ b/Makefile
@@ -263,6 +263,14 @@ ifeq ($(ARCH),x86_64)
         SRCARCH := x86
 endif
 
+# Additional ARCH settings for RISC-V
+ifeq ($(ARCH),riscv32)
+       SRCARCH := riscv
+endif
+ifeq ($(ARCH),riscv64)
+       SRCARCH := riscv
+endif
+
 # Additional ARCH settings for sparc
 ifeq ($(ARCH),sparc32)
        SRCARCH := sparc
````

This isn't so important because you can always just "./configure --build=riscv64-unknown-linux-gnu", but it is a bit of a headache...

sirnewton01, 2015-08-27:
+1

I'm interested in building a debian port. It'll be important to have consistent naming across the board.

manuelafm, 2016-04-28:
It would be quite nice if this could be resolved, for the reasons pointed out by Palmer.

terpstra, 2016-12-01:
Hi. I am not 100% clear on the state of this repository. I just got bit by this issue at the RISC-V workshop. Is there someone actively maintaining this repository? If not, I can try to merge some of this stuff in.

aswaterman, 2016-12-01:
I've only been passively maintaining it.  I'd appreciate the help.

a0u, 2016-12-02:
PR #48 fixes the immediate issue with the `uts.machine` field without needing to modify the top-level Makefile.

PR #48 (2016-12-02): riscv: Properly set UTS_MACHINE
This partially resolves #17.
closed 2016-12-02 by a0u