riscvarchive/riscv-linux #28: SMP kernel hangs at init under spike

jleidel, 2015-12-03:
I built an SMP kernel (default w/ 8 cpu support) from the riscv-linux master as per the instructions on the riscv website (886e5f99498baa2c4d1c1e3e1db801e6f4ae68c1).  I built the kernel using the riscv-tools master with the default configs.  I've tested with two different image files.  One built from scratch using busybox w/ a barebones config.  The other was built from my previous yocto build.  Both fail hang at the same point.  I've verified that both boot fine with the default 1 cpu.  

When ctrl-c'ing spike, it appears that cpu 0 and 1 are in infinite loops busy waiting for something.  The busywait appears to be somewhere near PC 0xffffffff8005bfa0.

jleidel, 2015-12-03:
quick update on this issue.  It appears the issue arises when building the SMP kernel support for 2 cpus and running spike with 2 cpus.

terpstra, 2016-12-03:
This should have been resolved by 030e8173dd9379de5c2f065ba73e734ff7647ef8.

closed 2016-12-03 by terpstra