riscvarchive/riscv-linux #34: Missing SFENCE.VM instructions

acw1251, 2016-05-05:
This port is currently missing some SFENCE.VM instructions according to version 1.7 of the supervisor spec. SFENCE.VM is needed to synchronize writes to page table entries through store instructions with later virtual address translations and hardware page table walks.

Right now SFENCE.VM is used in the `local_flush_tlb_all` macro and various other `flush_tlb` macros, but these macros aren't used when a previously invalid page table entry is updated to valid (for example see line 2602 in mm/memory.c in the full linux kernel).

The omitted SFENCE.VM instructions don't cause a bug in rocket or spike because their hardware page table walks are done in coherent memory (the required synchronization provided by SFENCE.VM in this case is a no-op). On the other hand, this omission does cause bugs in a single core processor with a write-back data cache and the hardware page table walk is performed through main memory. In this case, the required synchronization is writing back dirty lines from the data cache to main memory so the next hardware page table walk can see the newly written page table entries.

We have a RISC-V implementation with a write back cache and a hardware page table walk through main memory, and we implement SFENCE.VM as a TLB flush combined with writing back dirty lines in the data cache. When I run the current version of Linux with the default configuration, the boot process stalls right after looking at the htif block device (our root filesystem). Our processor diverges from spike with an unexpected page fault, and it seems like it hits an infinite loop of page faults.

To fix this, I added SFENCE.VM to the three function definitions below:
```
diff --git a/arch/riscv/include/asm/cacheflush.h b/arch/riscv/include/asm/cacheflush.h
index 56978bc..4cf9efe 100644
--- a/arch/riscv/include/asm/cacheflush.h
+++ b/arch/riscv/include/asm/cacheflush.h
@@ -11,6 +11,18 @@ static inline void local_flush_icache_all(void)
        asm volatile ("fence.i" ::: "memory");
 }
 
+#undef flush_cache_vmap
+static inline void flush_cache_vmap(unsigned long start, unsigned long end)
+{
+       asm volatile ("sfence.vm" ::: "memory");
+}
+
+#undef flush_cache_vunmap
+static inline void flush_cache_vunmap(unsigned long start, unsigned long end)
+{
+       asm volatile ("sfence.vm" ::: "memory");
+}
+
 #ifndef CONFIG_SMP
 
 #define flush_icache_range(start, end) local_flush_icache_all()
diff --git a/arch/riscv/include/asm/pgtable.h b/arch/riscv/include/asm/pgtable.h
index 74c1a29..a5a4a3e 100644
--- a/arch/riscv/include/asm/pgtable.h
+++ b/arch/riscv/include/asm/pgtable.h
@@ -265,6 +265,7 @@ static inline pte_t pte_modify(pte_t pte, pgprot_t newprot)
 static inline void update_mmu_cache(struct vm_area_struct *vma,
        unsigned long address, pte_t *ptep)
 {
+       asm volatile ("sfence.vm" ::: "memory");
 }
 
 /*
```

I don't know how good of a solution this is. This may be adding more SFENCE.VM instructions than necessary, but my main concern for this initial fix was that it was a sufficient number of them.

For more information, there was a long discussion on the RISC-V ISA Dev mailing list about SFENCE.VM semantics:
https://groups.google.com/a/groups.riscv.org/forum/#!topic/isa-dev/jbfnJSVLcWI

aswaterman, 2016-05-05:
Another approach (as discussed on that mailing list thread) is to allow spurious page faults in this case, and execute SFENCE.VM in the spurious fault handler.

acw1251, 2016-05-11:
That sounds like a good solution, but I did some digging, and I can't figure out where it would go. Right now it looks like spurious page faults (at least those detected in ```handle_pte_fault```) call the ```flush_tlb_fix_spurious_fault``` macro. RISC-V doesn't have a definition for that macro, so it looks like it falls back to the asm-generic definition which is ```flush_tlb_page``` but that is already defined as ```SFENCE.VM```. Maybe the spurious page fault is handled somewhere else?

If you have any pointers for places I should look, please let me know. Once I get around to it, I'll dump a trace of our processor executing a spurious page fault to see what code it is running.

aswaterman, 2016-05-11:
Thanks for looking into it, Andy.  It also seems to me that `flush_tlb_fix_spurious_fault` isn't called in all places we'd need for RISC-V, and so we'd presumably need to insert it in our own page fault handler.  Ultimately, that's probably the right approach.  I'd like @a0u to chime in on whether we should merge your patch in the mean time or handle spurious faults right now.  (He wrote the code, after all :-))

palmer-dabbelt, 2017-11-16:
I'm going to close this under the assumption that it's fixed.  The port has changed a lot in the last year and a half, so I'm not sure it applies any more.  If this is still an issue, feel free to either re-open it or submit a patch.

closed 2017-11-16 by palmer-dabbelt