riscvarchive/riscv-linux #38: Compilation problem in priv-1.9: conflicting types for 'atomic_or'

manuelafm, 2016-08-13:
```
  CC      arch/riscv/kernel/asm-offsets.s
In file included from include/linux/spinlock.h:416:0,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
include/linux/atomic.h:115:20: error: conflicting types for 'atomic_or'
 static inline void atomic_or(int i, atomic_t *v)
                    ^~~~~~~~~
In file included from include/linux/atomic.h:4:0,
                 from include/linux/spinlock.h:416,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
./arch/riscv/include/asm/atomic.h:239:20: note: previous definition of 'atomic_or' was here
 static inline void atomic_or(unsigned int mask, atomic_t *v)
                    ^~~~~~~~~
Kbuild:61: recipe for target 'arch/riscv/kernel/asm-offsets.s' failed
make[1]: *** [arch/riscv/kernel/asm-offsets.s] Error 1
Makefile:991: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
```

aswaterman, 2016-08-15:
Looks like you're using an old kernel version -- that conflicting
declaration in include/linux/atomic.h was removed in 2014.

manuelafm, 2016-08-15:
Oh, I see the problem now, thanks.

The reason why I was using 4.1 is because the README.rd / front page of the riscv-linux repository says that "full kernel source trees are maintained on separate branches tracking linux-stable" (mentioning 4.1 and 3.14 in particular), which I guess that it means LTS (long-term support) Linux releases.  So I was using the latest bugfix in the 4.1 series.

Which one should I use instead?  4.4 (was also a LTS, I think), or the 4.6.2 mentioned recently in a commit?

---

BTW, about the current problem, in the case that somebody else stumbles upon this searching in the internets...  The declaration is there in the upstream Kernel for 4.1 [1] from mid 2015, although it seems that it was removed later.  Locally:

```
$ grep 'void atomic_or(' include/linux/atomic.h arch/riscv/include/asm/atomic.h
include/linux/atomic.h:static inline void atomic_or(int i, atomic_t *v)
arch/riscv/include/asm/atomic.h:static inline void atomic_or(unsigned int mask, atomic_t *v)

$ cat ./include/config/kernel.release
4.1.26
```

[1] https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/include/linux/atomic.h?h=v4.1

manuelafm, 2016-08-22:
I used 4.6.6 successfully, so I think that this can be closed (doing it now).

closed 2016-08-22 by manuelafm