riscvarchive/riscv-linux #39: error: implicit declaration of function 'atomic64_xor'

harriss95, 2016-08-23:
> In file included from include/linux/spinlock.h:406:0,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
include/linux/atomic.h: In function 'atomic_andnot':
include/linux/atomic.h:464:2: error: implicit declaration of function 'atomic_and' [-Werror=implicit-function-declaration]
  atomic_and(~i, v);
  ^~~~~~~~~~
include/linux/atomic.h: At top level:
include/linux/atomic.h:468:33: error: redefinition of 'atomic_clear_mask'
 static inline __deprecated void atomic_clear_mask(unsigned int mask, atomic_t *v)
                                 ^~~~~~~~~~~~~~~~~
In file included from include/linux/atomic.h:4:0,
                 from include/linux/spinlock.h:406,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
./arch/riscv/include/asm/atomic.h:224:20: note: previous definition of 'atomic_clear_mask' was here
 static inline void atomic_clear_mask(unsigned int mask, atomic_t *v)
                    ^~~~~~~~~~~~~~~~~
In file included from include/linux/spinlock.h:406:0,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
include/linux/atomic.h:473:33: error: redefinition of 'atomic_set_mask'
 static inline __deprecated void atomic_set_mask(unsigned int mask, atomic_t *v)
                                 ^~~~~~~~~~~~~~~
In file included from include/linux/atomic.h:4:0,
                 from include/linux/spinlock.h:406,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
./arch/riscv/include/asm/atomic.h:239:20: note: previous definition of 'atomic_set_mask' was here
 static inline void atomic_set_mask(unsigned int mask, atomic_t *v)
                    ^~~~~~~~~~~~~~~
In file included from include/linux/spinlock.h:406:0,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
include/linux/atomic.h: In function 'atomic_set_mask':
include/linux/atomic.h:475:2: error: implicit declaration of function 'atomic_or' [-Werror=implicit-function-declaration]
  atomic_or(mask, v);
  ^~~~~~~~~
include/linux/atomic.h: In function 'atomic64_andnot':
include/linux/atomic.h:589:2: error: implicit declaration of function 'atomic64_and' [-Werror=implicit-function-declaration]
  atomic64_and(~i, v);
  ^~~~~~~~~~~~
In file included from include/linux/atomic.h:593:0,
                 from include/linux/spinlock.h:406,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/sched.h:19,
                 from arch/riscv/kernel/asm-offsets.c:2:
include/asm-generic/atomic-long.h: In function 'atomic_long_or':
include/asm-generic/atomic-long.h:26:28: error: implicit declaration of function 'atomic64_or' [-Werror=implicit-function-declaration]
 #define ATOMIC_LONG_PFX(x) atomic64 ## x
                            ^
include/asm-generic/atomic-long.h:121:2: note: in expansion of macro 'ATOMIC_LONG_PFX'
  ATOMIC_LONG_PFX(_##op)(i, v);     \
  ^~~~~~~~~~~~~~~
include/asm-generic/atomic-long.h:127:1: note: in expansion of macro 'ATOMIC_LONG_OP'
 ATOMIC_LONG_OP(or)
 ^~~~~~~~~~~~~~
include/asm-generic/atomic-long.h: In function 'atomic_long_xor':
include/asm-generic/atomic-long.h:26:28: error: implicit declaration of function 'atomic64_xor' [-Werror=implicit-function-declaration]
 #define ATOMIC_LONG_PFX(x) atomic64 ## x
                            ^
include/asm-generic/atomic-long.h:121:2: note: in expansion of macro 'ATOMIC_LONG_PFX'
  ATOMIC_LONG_PFX(_##op)(i, v);     \
  ^~~~~~~~~~~~~~~
include/asm-generic/atomic-long.h:128:1: note: in expansion of macro 'ATOMIC_LONG_OP'
 ATOMIC_LONG_OP(xor)
 ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
Kbuild:80: recipe for target 'arch/riscv/kernel/asm-offsets.s' failed
make[1]: *** [arch/riscv/kernel/asm-offsets.s] Error 1
Makefile:1006: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2

terpstra, 2016-08-23:
That sounds like you're using the wrong kernel version with the repository.
Which branch and which kernel?

harriss95, 2016-08-23:
im using 4.6.2 and priv-1.9 branch

terpstra, 2016-08-23:
Please run:

curl -L https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.6.2.tar.xz |
tar -xJ

cd linux-4.6.2

git init

git remote add -t priv-1.9 origin https://github.com/riscv/riscv-linux.git

git fetch

git checkout -f -t origin/priv-1.9

make ARCH=riscv defconfig

make ARCH=riscv

... because I cannot reproduce this problem with 4.6.2. Does the problem
persist?

harriss95, 2016-08-23:
Problem solved . Thank you 

closing the issue

closed 2016-08-23 by harriss95