riscvarchive/riscv-linux #42: Multicore (SMP) support?

aneesh-neelam, 2016-11-21:
I've been trying to build and run the kernel on a multiple core (32-1024 cores) machine, using [Spike ISA Simulator](https://github.com/riscv/riscv-isa-sim). I tried to run that using [QEMU](https://github.com/riscv/riscv-qemu) as well, but that did not work as the machine_type did not support an SMP greater than 1. 

Here is the console output I get when I run it using [Spike ISA Simulator](https://github.com/riscv/riscv-isa-sim): 
```
aneeshneelam@Aneesh-Ubuntu-VirtualBox:~/Developer/test$ spike +disk=root.bin bbl vmlinux -p 1024
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] Linux version 4.1.35-g174f395-dirty (aneeshneelam@Aneesh-Ubuntu-VirtualBox) (gcc version 5.3.0 (GCC) ) #3 SMP Mon Oct 31 14:39:43 PDT 2016
[    0.000000] Available physical memory: 2044MB
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000200000-0x000000007fdfffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000007fdfffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000007fdfffff]
[    0.000000] PERCPU: Embedded 12 pages/cpu @ffffffff81ef0000 s16888 r0 d32264 u49152
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 516110
[    0.000000] Kernel command line: root=/dev/htifblk0
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2058284K/2093056K available (2031K kernel code, 97K rwdata, 416K rodata, 92K init, 224K bss, 34772K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=32 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS:2
[    0.000000] clocksource riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627 ns
[    0.150000] Calibrating delay using timer specific routine.. 20.01 BogoMIPS (lpj=100050)
[    0.150000] pid_max: default: 32768 minimum: 301
[    0.150000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.150000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.150000] Brought up 1 CPUs
[    0.150000] devtmpfs: initialized
[    0.150000] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.150000] NET: Registered protocol family 16
[    0.150000] Switched to clocksource riscv_clocksource
[    0.150000] NET: Registered protocol family 2
[    0.150000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.150000] TCP bind hash table entries: 16384 (order: 6, 262144 bytes)
[    0.150000] TCP: Hash tables configured (established 16384 bind 16384)
[    0.150000] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.150000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.150000] NET: Registered protocol family 1
[    0.150000] futex hash table entries: 256 (order: 1, 8192 bytes)
[    0.150000] io scheduler noop registered
[    0.150000] io scheduler cfq registered (default)
[    0.160000] htifcon htif1: detected console
[    0.160000] console [htifcon0] enabled
[    0.160000] htifblk htif2: detected disk
[    0.160000] htifblk htif2: added htifblk0
[    0.160000] VFS: Mounted root (ext2 filesystem) readonly on device 254:0.
[    0.160000] devtmpfs: mounted
[    0.160000] Freeing unused kernel memory: 92K (ffffffff80000000 - ffffffff80017000)
[    0.170000] EXT2-fs (htifblk0): warning: mounting unchecked fs, running e2fsck is recommended
# cat /proc/cpuinfo 
hart	: 0
isa	: RV64G
```

I did enable SMP support and modify the max number of CPU cores in the kernel config before building it. 

I can see that RCU restricts the number of CPUs to 1, I'm researching what that means currently. However, am I missing something when I'm building the kernel?

jleidel, 2016-11-25:
aneesh, there are a few issues with your command line parameters for Spike.  First, the syntax for enabling SMP support using spike is: "-pN", not "-p N".  Note the lack of a space between the argument and the N cpus you want.  Also note that there is a limit to the number of cpus currently supported with the RISC-V Linux port.  At one time it was 8 (even if you build the kernel with something larger).  I believe this is still the case.

So, your command line should be: 
spike -p8 +disk=root.bin bbl vmlinux

aneesh-neelam, 2016-11-26:
That was helpful, thanks. 

Can you tell me what is the maximum that [Spike ISA Simulator](https://github.com/riscv/riscv-isa-sim) supports?

jleidel, 2016-11-28:
aneesh, I believe its currently 8.  Nothing larger seems to boot based Init.  I pinged the riscv-sw mailing list.  no response yet.

aneesh-neelam, 2016-11-30:
Very well, thank you.

terpstra, 2016-12-03:
Has this issue been resolved?

palmer-dabbelt, 2017-05-06:
I'm going to close this under the assumption it's dead: if it's still a problem then please re-open the bug.

closed 2017-05-06 by palmer-dabbelt