riscvarchive/riscv-linux #44: riscv configuration cannot work on 32bit OS

shrshore, 2016-11-28:
Are there any running configuration file can be shared???Or how should I modify the configuration to make it work? 
The riscv configuration file just cannot work on 32bit OS. There are so many errors, such as
```
In file included from arch/riscv/include/generated/asm/div64.h:1:0,
                 from include/linux/kernel.h:136,
                 from include/linux/list.h:8,
                 from include/linux/preempt.h:10,
                 from include/linux/spinlock.h:50,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/uapi/linux/timex.h:56,
                 from include/linux/timex.h:56,
                 from include/linux/clocksource.h:12,
                 from arch/riscv/kernel/time.c:1:
include/asm-generic/div64.h:224:22: error: passing argument 1 of '__div64_32' from incompatible pointer type [-Werror=incompatible-pointer-types]
   __rem = __div64_32(&(n), __base); \
                      ^
arch/riscv/kernel/time.c:81:2: note: in expansion of macro 'do_div'
  do_div(lpj_fine, HZ);
  ^~~~~~
```

the OS is 
```
Linux ubuntu 4.4.0-47-generic #68-Ubuntu SMP Wed Oct 26 19:39:59 UTC 2016 i686 i686 i686 GNU/Linux
```
And I try these but only errors are found.
```
make ARCH=riscv defconfig
make ARCH=riscv menuconfig
make -j4 ARCH=riscv vmlinux
```

terpstra, 2016-12-02:
What version of risc-v tools are you using and which kernel version? It is very unlikely that error is caused by your host CPU being 32-bits. The file being compiled in your example was a RISC-V file.

sorear, 2016-12-16:
@terpstra This appears to have been an attempt to compile the kernel for riscv32.

palmer-dabbelt, 2017-05-06:
This has been fixed.

closed 2017-05-06 by palmer-dabbelt

s-macke, 2017-05-07:
Check this emulator:
http://s-macke.github.io/jor1k/demos/riscv.html
RISC-V 32-Bit is finally working with the new privilege spec.

terpstra, 2017-05-07:
Very cool! I see you copied spike's device tree string, but changed mmu-type to sv32. Does this mean spike now also can run linux 32-bit? Did you have to change much in the kernel to handle sv32? Are those changes merged?

s-macke, 2017-05-07:
Following tree should compile out of the box.
https://github.com/riscv/riscv-linux/tree/riscv-next

Today,  I have also commented on the necessary changes for spike here: https://github.com/riscv/riscv-linux/issues/61

Not much to do I think. It is just the console handling.