riscvarchive/riscv-linux #50: OS support for virtual memory

susi655, 2017-01-04:
What kind of support is provided by RISCV Linux OS for virtual memory?

For example,
Is page walk managed by OS?
What about TLB filling? etc.

sorear, 2017-01-04:
Linux uses the standard page table described in the privileged spec.  How TLBs work is a question for your chip vendor.

susi655, 2017-01-05:
So TLB filling is done by hardware. Is it?

sorear, 2017-01-05:
It can be done by hardware or software.  It is _not_ done by Linux.

susi655, 2017-01-05:
Few doubts regarding Page table
   Is it optional to implement hardware page walker?
   If it is can be done by software, up to what extent OS will do?
   Means decision of which page to replace will done by OS?
Is OS looking for multi-level paging?

sorear, 2017-01-05:
The bug tracker is not a forum and this isn't even a related repository.

Page table walker is not in the OS.  If it's software, it's non-OS software (e.g. M-mode trap handled in a M-mode monitor program that runs above Linux).  Why do you keep asking?  What are you trying to achieve?

*As far as any OS knows*, page tables are walked automatically.

palmer-dabbelt, 2017-05-06:
I'm closing this: like @sorear said this isn't the right place to ask this question.  isa-dev@groups.riscv.org might be a better place to do it.

closed 2017-05-06 by palmer-dabbelt