riscvarchive/riscv-linux #56: Error while building kernel on CentOS 7.3

sinkswim, 2017-03-10:
While running 'make ARCH=riscv'  on CentOS I get the following:
```
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CC      kernel/bounds.s
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-m64'
make[1]: *** [kernel/bounds.s] Error 1
make: *** [prepare0] Error 2
```

palmer-dabbelt, 2017-03-13:
The GCC flags have changed, and I think you're on an old branch (master is old now, we're in a bit of a transition to tracking upstream).  Can you try "riscv-next" or "priv-1.10"?  I see

````
$ git grep -- "-march" | grep riscv
arch/riscv/Makefile:KBUILD_AFLAGS += -march=$(KBUILD_MARCH)$(KBUILD_RV_ATOMIC)fd$(KBUILD_RVC)
arch/riscv/Makefile:KBUILD_CFLAGS += -march=$(KBUILD_MARCH)$(KBUILD_RV_ATOMIC)$(KBUILD_RVC)
````

so there shouldn't be any "-m64" any more.

sinkswim, 2017-03-13:
Resolved, thanks.

closed 2017-03-13 by sinkswim

yonghaoXuGit, 2018-09-16:
Hi，I encountered the same problem.How did you solve it？Can you tell me？

Thanks！