riscvarchive/riscv-linux #60: Corrupt SV32 page table generation in the function setup_vm

s-macke, 2017-04-23:
Since commit e84bcd9070b6765e9bea0308dde86bcb5b3ea5a8 the setup of the page table is broken for SV32 type page tables. 
The function
https://github.com/riscv/riscv-linux/blob/priv-1.10/arch/riscv/kernel/setup.c#L78
assumes a Page Middle Directory (3 level page table) with is not available for SV32.

I have hacked a solution, which does work for me and 32-Bit Linux boots again. You might want to use is as basis for a fix. But the code ignores the other page table types.

```
asmlinkage void __init setup_vm(void)
{
        extern char _start;
        uintptr_t i;
        uintptr_t pa = (uintptr_t) &_start;
        uintptr_t prot = pgprot_val(PAGE_KERNEL) | _PAGE_EXEC;
        pgd_t pmd = __pgd((PFN_DOWN(pa) << _PAGE_PFN_SHIFT) | prot);
        va_pa_offset = PAGE_OFFSET - pa;
        pfn_base = PFN_DOWN(pa);
        trampoline_pg_dir[(PAGE_OFFSET >> PGDIR_SHIFT) % PTRS_PER_PGD] = pmd;
        for (i = 0; i < (-PAGE_OFFSET)/PGDIR_SIZE; ++i) {
                swapper_pg_dir[(PAGE_OFFSET >> PGDIR_SHIFT) % PTRS_PER_PGD + i] =
                        __pgd((PFN_DOWN(pa) << _PAGE_PFN_SHIFT) | prot);
                pa += PGDIR_SIZE;
        }
}
```

aswaterman, 2017-04-23:
Thanks, Sebastian.  Can you verify that 26e4369ab1f94f32cdec0a53c0193565c3f6deb4 works correctly for RV32?

s-macke, 2017-04-24:
The patch is almost fine except that the PMD_SIZE is defined in asm-generic/pgtable-nopmd.h if no pmd is present. That was the main problem with the code. The code compiled and ran on 32-Bit but lead to a corrupt pagetable. The correct way is to use __PAGETABLE_PMD_FOLDED. This define is also used in arch/riscv/include/asm/pgalloc.h

```
index b7981d7065ea..2a6ce882081a 100644
--- a/arch/riscv/kernel/setup.c
+++ b/arch/riscv/kernel/setup.c
@@ -91,7 +91,7 @@ asmlinkage void __init setup_vm(void)
        BUG_ON((PAGE_OFFSET % PGDIR_SIZE) != 0);
        BUG_ON((pa % (PAGE_SIZE * PTRS_PER_PTE)) != 0);

-#ifdef PMD_SIZE
+#ifndef __PAGETABLE_PMD_FOLDED
        trampoline_pg_dir[(PAGE_OFFSET >> PGDIR_SHIFT) % PTRS_PER_PGD] =
                pfn_pgd(PFN_DOWN((uintptr_t)trampoline_pmd),
                        __pgprot(_PAGE_TABLE));
```

aswaterman, 2017-04-24:
Ah, thanks, fixed in 8fa2c8f57ef9ddac4dec0965f5849d0f004b2337.

closed 2017-04-24 by aswaterman