riscvarchive/riscv-linux #63: Stack alignment broken in Linux

s-macke, 2017-05-07:
There is still a way to get an unaligned stack pointer in the Linux kernel under RV32. The patch shows the problem. However, you might want to correct this with compatibility under 64-Bit :smiley:

```
--- a/arch/riscv/kernel/entry.S
+++ b/arch/riscv/kernel/entry.S
@@ -37,7 +37,7 @@
 _restore_kernel_sp:
        csrr sp, sscratch
 _save_context:
-       addi sp, sp, -(PT_SIZE)
+       addi sp, sp, -144
        REG_S x1,  PT_RA(sp)
        REG_S x3,  PT_GP(sp)
        REG_S x4,  PT_TP(sp)
@@ -211,7 +211,7 @@ resume_userspace:
        bnez s1, work_pending

        /* Save unwound kernel stack pointer in sscratch */
-       addi s0, sp, PT_SIZE
+       addi s0, sp, 144
        csrw sscratch, s0
 restore_all:
        RESTORE_ALL
```

aswaterman, 2017-05-07:
@palmer-dabbelt can you fix?  It should be -ALIGN(PT_SIZE, STACK_ALIGN).

s-macke, 2017-05-08:
Unfortunately no: 
```
  AS      arch/riscv/kernel/entry.o
arch/riscv/kernel/entry.S: Assembler messages:
arch/riscv/kernel/entry.S:129: Error: illegal operands `addi sp,sp,-.balign 4(140,STACK_ALIGN)'
arch/riscv/kernel/entry.S:214: Error: illegal operands `addi s0,sp,.balign 4(140,STACK_ALIGN)'
```
Also it didn't work when I defined STACK_ALIGN via #include <asm/processor.h>

My guess is, that you need another logic for the assembler than for C. I have searched a few other assembler files but couldn't find an alignment function we need here.

palmer-dabbelt, 2017-06-03:
Sorry, I seem to have dropped this.  I believe it's fixed by https://github.com/riscv/riscv-linux/commit/7fd16e63266c9213888e8a1105836e834052a128

closed 2017-06-03 by palmer-dabbelt