riscvarchive/riscv-linux #66: get_cycles() function name a misnomer.

ccelio, 2017-05-23:
Hey guys,

I'm not sure if this breaks a convention on Linux's side, but in trying to read the timing code, I became confused because `get_cycles()` is actually reading the `time` CSR and not the `cycles` CSR:

https://github.com/riscv/riscv-linux/blob/riscv-next/arch/riscv/kernel/time.c#L51
https://github.com/riscv/riscv-linux/blob/riscv-next/arch/riscv/include/asm/timex.h#L24

Likewise `cycles_t`:
https://github.com/riscv/riscv-linux/blob/riscv-next/arch/riscv/include/asm/timex.h#L22

aswaterman, 2017-05-23:
We followed the example of x86, which implements ```get_cycles``` using ```rdtsc```.  Like the ```time``` CSR, ```rdtsc``` tracks real time, not clock cycles.

palmer-dabbelt, 2017-05-23:
Ya, it's super odd.  This was on my code review list for Linux, and I determined this was the correct thing to do.

closed 2017-05-23 by palmer-dabbelt