riscvarchive/riscv-linux #68: Does linux 4.6.2 support SMP well？

yaoluker, 2017-05-27:
Hi，
Recently，I generated a 4 Rocket core RISCV project，and changed the menuconfig to support multicore.
But when the system boot up, it only recognized one core.
I'm wondering if there is any example to boot up linux for multiple cores? Or, is there anyone who has realized  multicore Linux boot could give me any advise？

Thanks！

YaoLiu

palmer-dabbelt, 2017-06-01:
I'm not sure what branch you're on, as there's a lot of them.  Here's the code that boots the CPUs other than the main hart

https://github.com/riscv/riscv-linux/blob/riscv-next/arch/riscv/kernel/head.S#L104
````
.Lsecondary_start:
#ifdef CONFIG_SMP
	li a1, CONFIG_NR_CPUS
	bgeu a0, a1, .Lsecondary_park

	la a1, __cpu_up_stack_pointer
	slli a0, a0, LGREG
	add a0, a0, a1

.Lwait_for_cpu_up:
	REG_L sp, (a0)
	beqz sp, .Lwait_for_cpu_up
	fence

	/* Enable virtual memory and relocate to virtual address */
	call relocate

	/* Initialize task_struct pointer */
	li tp, -THREAD_SIZE
	add tp, tp, sp

	tail smp_callin
#endif
````

if you don't have that then you're not going to have the other CPUs boot.

noamc, 2017-06-05:
What is the difference between core and hart?
Is hart for riscv is same as hyper thread for x86?

TNX

austinharris, 2017-06-16:
riscv-next seems to not be able to boot hart 0 (spike -p1 doesn't boot, spike -p2 shows only hart 1 in cpuinfo.) riscv-for-submission-v2 fails to compile for me with SMP enabled (raw_smp_processor_id() tries to use the cpu field which appears to have been removed from the thread_info struct.) Is the SMP support in flux until the upstreaming process is finished?

palmer-dabbelt, 2017-11-16:
I'm closing this: we recently updated Freedom and the SDKs, so there's really no reason to use the old kernels any more.  We recently landed upstream, anything before that is almost certainly obscenely buggy :).

closed 2017-11-16 by palmer-dabbelt