riscvarchive/riscv-linux #72: Having booting kernel for buildroot

noamc, 2017-06-15:
Hi 
I managed to create buildroot toolchiain for riscv64 and I use riscv-next branch of Linux.
however unlike the master branch of riscv-linux the riscv-next branch fails to boot (bbl silently hangs).
I wish to use master, but it is only having the riscv arch directory.
The master readme file says [Full kernel source trees](https://github.com/riscv/riscv-linux/tree/master#full-kernel-source-trees), but I can't find any linux-4.6.y-riscv  branch.

Please assist

palmer-dabbelt, 2017-06-16:
Can you try the "riscv-for-submission-v2" branch?  The Linux repo is in flux right now as we're in the process of upstreaming, hopefully this will all be cleaned up soon.  Sorry for the trouble!

BenjaminLpn, 2017-08-09:
@palmer-dabbelt  I encountered this problem from the latest riscv-next. Please tell me whether riscv-next is updated. Thanks.

palmer-dabbelt, 2017-11-16:
freedom-u-sdk should have everything matching now

closed 2017-11-16 by palmer-dabbelt