riscvarchive/riscv-linux #76: Any todo list for upstream?

poweihuang17, 2017-06-29:
Hi,
I saw a lot of review and submit in the patch mailing list.
Could I know the todo list so that i could pick up some corner to help?
Thanks

palmer-dabbelt, 2017-06-29:
Here's our latest patch set <https://lkml.org/lkml/2017/6/28/734> email submission.  That contains the entire set of code reviews that currently need to be looked at.  I generally go through them in the order I get the messages back, which means I'm currently looking at <https://www.spinics.net/lists/kernel/msg2545129.html>.

I just hopped into #riscv on freenode if you want to sync up more closely, as email can be a bit slow.

poweihuang17, 2017-07-24:
@palmer-dabbelt Thanks. It seems that we might not be able to catch up 4.13...? Do you have some plan or need some help for 4.14 upstream? While I don't have ability to help, some taiwanese linux kernel hackers are pretty interested in it.

poweihuang17, 2017-07-24:
Overall, thanks for your effort for linux upstream.

palmer-dabbelt, 2017-07-24:
Ya, we've missed 4.13.  If you've got some people who are interested then there's a bit of work to do, let's sync up over email.

poweihuang17, 2017-07-24:
OK. Let's discuss this in patch mailing list.

farhan90, 2017-10-05:
Hi, 
I would be interested in contributing to Linux patches and even QEMU for risc v.

palmer-dabbelt, 2017-10-06:
@farhan90 Great!  Right now I think the biggest issue we have is QEMU, as our Linux port is in decent shape (we hope to be in linux-next soon).  If you're familiar with QEMU then it would be great to have some time spent on it.  Obviously volunteer efforts are appreciated, but if you need some financial support then feel free to email me and we can talk.

Right now the biggest issue with QEMU is that it's not up to date with the latest privileged specification.  If you can get QEMU booting the latest Linux kernels, that would go a long way towards enabling people to start contributing to the various RISC-V ports.  I though there was a pull request <https://github.com/riscv/riscv-qemu/pulls> floating around somewhere to update to 1.10, but I can't find it any more.

Additionally, I just sent a request for help running a continuous integration server to the RISC-V software mailing lists.  Having a working QEMU would greatly help that effort.

farhan90, 2017-10-06:
hey Palmer,

Thanks for the response. I am somewhat familiar with qemu as I work professionally with qemu
https://git.qemu.org/qemu.git/?p=qemu.git&a=search&h=HEAD&st=author&s=Farhan+Ali

I have also worked on the Linux kernel for s390x architecture.

I do not need any financial support and I am mostly interested in working on it as a hobby and volunteer since recently I have been learning and taking deeper interest in computer architectures.  I am still trying to learn more about the RISC-V architecture and ISA.

palmer-dabbelt, 2017-10-06:
Great!  If you have any questions, feel free to ask!  I think working on QEMU is a great way to learn an ISA, you'll really have to understand the whole ISA if you have to emulate it :).

The sw-dev@groups.riscv.org is the best place to ask these sorts of questions.  Feel free to CC me on anything if you need help.

farhan90, 2017-10-06:
Yup :) 

I definitely want to get involved.  For QEMU, I know there is a 2.10 release and so we probably need to start by rebasing the branch to the latest master and see.

palmer-dabbelt, 2017-10-06:
Great, thanks!  There's a handful of people involved in the RISC-V QEMU port, so you might want to ping them so they don't get grumpy, but progress has been slow so I'm sure they'll be happy to have help.  I agree bumping to the latest upstream is the first thing to do, as otherwise it's hard to reason about a port.

I think you should move this discussion to either the QEMU issue tracker or the RISC-V mailing lists, as the Linux issue tracker won't get the right people to see your messages.  I'm going to close this bug, but feel free to CC or @ me on wherever you end up next.

Thanks!

closed 2017-10-06 by palmer-dabbelt

tommythorn, 2017-11-22:
Is QEMU so important?  RISCVEMU (https://bellard.org/riscvemu/) is pretty
fast and boots Linux (with X11 support no less).  That said, I haven't
tested
how well the latest versions of the Linux kernel runs on it.

Tommy