riscvarchive/riscv-linux #77: ERROR while building vmlinux

shiningjg, 2017-06-30:
Hello,
I meet some errors when making vmlinux.
`[jg@localhost riscv-linux]$ make CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv64 -j4
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  WRAP    arch/riscv/include/generated/asm/bugs.h
  WRAP    arch/riscv/include/generated/asm/cacheflush.h
  WRAP    arch/riscv/include/generated/asm/checksum.h
  WRAP    arch/riscv/include/generated/asm/clkdev.h
  WRAP    arch/riscv/include/generated/asm/current.h
  WRAP    arch/riscv/include/generated/asm/cputime.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  WRAP    arch/riscv/include/generated/asm/div64.h
  WRAP    arch/riscv/include/generated/asm/dma.h
  WRAP    arch/riscv/include/generated/asm/emergency-restart.h
  WRAP    arch/riscv/include/generated/asm/errno.h
  WRAP    arch/riscv/include/generated/asm/exec.h
  WRAP    arch/riscv/include/generated/asm/fb.h
  WRAP    arch/riscv/include/generated/asm/fcntl.h
  WRAP    arch/riscv/include/generated/asm/ftrace.h
  WRAP    arch/riscv/include/generated/asm/futex.h
  WRAP    arch/riscv/include/generated/asm/hardirq.h
  WRAP    arch/riscv/include/generated/asm/hash.h
  WRAP    arch/riscv/include/generated/asm/hw_irq.h
  WRAP    arch/riscv/include/generated/asm/ioctl.h
  WRAP    arch/riscv/include/generated/asm/ioctls.h
  WRAP    arch/riscv/include/generated/asm/ipcbuf.h
  WRAP    arch/riscv/include/generated/asm/irq_regs.h
  WRAP    arch/riscv/include/generated/asm/irq_work.h
  WRAP    arch/riscv/include/generated/asm/kdebug.h
  WRAP    arch/riscv/include/generated/asm/kmap_types.h
  WRAP    arch/riscv/include/generated/asm/local.h
  WRAP    arch/riscv/include/generated/asm/kvm_para.h
  WRAP    arch/riscv/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/riscv/include/generated/asm/mman.h
  WRAP    arch/riscv/include/generated/asm/module.h
  WRAP    arch/riscv/include/generated/asm/msgbuf.h
  WRAP    arch/riscv/include/generated/asm/mutex.h
  WRAP    arch/riscv/include/generated/asm/param.h
  WRAP    arch/riscv/include/generated/asm/percpu.h
  WRAP    arch/riscv/include/generated/asm/poll.h
  WRAP    arch/riscv/include/generated/asm/posix_types.h
  WRAP    arch/riscv/include/generated/asm/preempt.h
  WRAP    arch/riscv/include/generated/asm/resource.h
  WRAP    arch/riscv/include/generated/asm/scatterlist.h
  WRAP    arch/riscv/include/generated/asm/sections.h
  WRAP    arch/riscv/include/generated/asm/sembuf.h
  WRAP    arch/riscv/include/generated/asm/shmbuf.h
  WRAP    arch/riscv/include/generated/asm/shmparam.h
  WRAP    arch/riscv/include/generated/asm/signal.h
  WRAP    arch/riscv/include/generated/asm/socket.h
  WRAP    arch/riscv/include/generated/asm/sockios.h
  WRAP    arch/riscv/include/generated/asm/stat.h
  WRAP    arch/riscv/include/generated/asm/statfs.h
  WRAP    arch/riscv/include/generated/asm/swab.h
  WRAP    arch/riscv/include/generated/asm/termbits.h
  WRAP    arch/riscv/include/generated/asm/termios.h
  WRAP    arch/riscv/include/generated/asm/topology.h
  WRAP    arch/riscv/include/generated/asm/trace_clock.h
  WRAP    arch/riscv/include/generated/asm/types.h
  WRAP    arch/riscv/include/generated/asm/user.h
  WRAP    arch/riscv/include/generated/asm/unaligned.h
  WRAP    arch/riscv/include/generated/asm/vga.h
  WRAP    arch/riscv/include/generated/asm/vmlinux.lds.h
  WRAP    arch/riscv/include/generated/asm/xor.h
  WRAP    arch/riscv/include/generated/uapi/asm/setup.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  CC      scripts/mod/empty.o
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-mstrict-align'; did you mean '-Wstrict-aliasing'?
make[2]: *** [scripts/mod/empty.o] Error 1
make[1]: *** [scripts/mod] Error 2
make[1]: *** Waiting for unfinished jobs....
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
scripts/dtc/checks.c: In function ‘check_simple_bus_reg’:
scripts/dtc/checks.c:876:2: warning: format ‘%zx’ expects argument of type ‘size_t’, but argument 4 has type ‘uint64_t’ [-Wformat=]
  snprintf(unit_addr, sizeof(unit_addr), "%zx", reg);
  ^
scripts/dtc/checks.c:876:2: warning: format ‘%zx’ expects argument of type ‘size_t’, but argument 4 has type ‘uint64_t’ [-Wformat=]
  HOSTCC  scripts/dtc/util.o
  SHIPPED scripts/dtc/dtc-lexer.lex.c
  SHIPPED scripts/dtc/dtc-parser.tab.h
  SHIPPED scripts/dtc/dtc-parser.tab.c
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
make: *** [scripts] Error 2
make: *** Waiting for unfinished jobs....
  UPD     include/config/kernel.release
`
It seems that riscv64-unknown-linux-gnu-gcc does not support the option '-mstrict-align', how to deal with it then?

aswaterman, 2017-06-30:
```-mstrict-align``` was introduced relatively recently (early 2017, I think), so you likely need to upgrade gcc.

palmer-dabbelt, 2017-07-03:
This should fix it.  It's part of the submission (the riscv-for-submission-v3 branch).

1cd2e072bbd1a518e11b551435be0ad84a2c52af

palmer-dabbelt, 2017-11-16:
I'm assuming this got fixed.

closed 2017-11-16 by palmer-dabbelt