riscvarchive/riscv-linux #81: Error command not found

Seonghun94, 2017-08-01:
First of All, I can't speak English well.. please understand me^^
I was execute : `sudo make -j16 ARCH=riscv  CROSS_COMPILE=riscv64-unknown-linux-gnu-`
result : 

> ./scripts/gcc-version.sh: line 25: riscv64-unknown-linux-gnu-gcc: command not found
./scripts/gcc-version.sh: line 26: riscv64-unknown-linux-gnu-gcc: command not found
make: riscv64-unknown-linux-gnu-gcc: Command not found
make: riscv64-unknown-linux-gnu-gcc: Command not found
scripts/kconfig/conf  --silentoldconfig Kconfig
./scripts/gcc-version.sh: line 25: riscv64-unknown-linux-gnu-gcc: command not found
./scripts/gcc-version.sh: line 26: riscv64-unknown-linux-gnu-gcc: command not found
make: riscv64-unknown-linux-gnu-gcc: Command not found
make: riscv64-unknown-linux-gnu-gcc: Command not found
  CHK     include/config/kernel.release
make: riscv64-unknown-linux-gnu-gcc: Command not found
make: riscv64-unknown-linux-gnu-gcc: Command not found
  CHK     include/generated/uapi/linux/version.h
make: riscv64-unknown-linux-gnu-gcc: Command not found
make: riscv64-unknown-linux-gnu-gcc: Command not found
  CC      scripts/mod/empty.o
/bin/sh: 1: riscv64-unknown-linux-gnu-gcc: not found
scripts/Makefile.build:302: recipe for target 'scripts/mod/empty.o' failed
make[2]: *** [scripts/mod/empty.o] Error 127
scripts/Makefile.build:561: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
Makefile:570: recipe for target 'scripts' failed
make: *** [scripts] Error 2
make: *** Waiting for unfinished jobs....

it seems that path of riscv64-unknown-linux-gnu-gcc does fault.
because I fixed my .bashrc file that can't read $PATH

kito-cheng, 2017-08-01:
Try again without `sudo`?

Seonghun94, 2017-08-01:
thanks for your reply
`make -j16 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-`

> CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
/bin/sh: 1: cannot create include/generated/uapi/linux/version.h.tmp: Permission denied
Makefile:1125: recipe for target 'include/generated/uapi/linux/version.h' failed
make: *** [include/generated/uapi/linux/version.h] Error 2
make: *** Waiting for unfinished jobs....
/bin/sh: 1: cannot create include/config/kernel.release.tmp: Permission denied
Makefile:1032: recipe for target 'include/config/kernel.release' failed
make: *** [include/config/kernel.release] Error 2
make: *** wait: No child processes.  Stop.

palmer-dabbelt, 2017-11-16:
I'm closing this as not-a-bug.

closed 2017-11-16 by palmer-dabbelt