riscvarchive/riscv-linux #82: make headers_install fail

kito-cheng, 2017-08-21:
```
$ git branch 
* riscv-for-submission-v7
$ make defconfig ARCH=riscv
...
*** Default configuration is based on 'spike64_defconfig'
#
# configuration written to .config
#
$ make headers_install ARCH=riscv
...
scripts/Makefile.headersinst:62: *** Some mandatory headers (errno.h fcntl.h ioctl.h ioctls.h ipcbuf.h mman.h msgbuf.h param.h poll.h posix_types.h resource.h sembuf.h shmbuf.h signal.h socket.h sockios.h stat.h statfs.h swab.h termbits.h termios.h types.h) are missing in arch/riscv/include/uapi/asm.  Stop.
scripts/Makefile.headersinst:31: recipe for target 'asm' failed
make[1]: *** [asm] Error 2
Makefile:1167: recipe for target 'headers_install' failed
make: *** [headers_install] Error 2
```

kito-cheng, 2017-08-21:
I just work around by add it into `arch/riscv/include/uapi/asm/Kbuild`, but I am not sure it's right or not :P
```diff
diff --git a/arch/riscv/include/uapi/asm/Kbuild b/arch/riscv/include/uapi/asm/Kbuild
index a20ed2c..5ded96b 100644
--- a/arch/riscv/include/uapi/asm/Kbuild
+++ b/arch/riscv/include/uapi/asm/Kbuild
@@ -3,3 +3,25 @@ include include/uapi/asm-generic/Kbuild.asm
 
 generic-y += setup.h
 generic-y += unistd.h
+generic-y += errno.h
+generic-y += fcntl.h
+generic-y += ioctl.h
+generic-y += ioctls.h
+generic-y += ipcbuf.h
+generic-y += mman.h
+generic-y += msgbuf.h
+generic-y += param.h
+generic-y += poll.h
+generic-y += posix_types.h
+generic-y += resource.h
+generic-y += sembuf.h
+generic-y += shmbuf.h
+generic-y += signal.h
+generic-y += socket.h
+generic-y += sockios.h
+generic-y += stat.h
+generic-y += statfs.h
+generic-y += swab.h
+generic-y += termbits.h
+generic-y += termios.h
+generic-y += types.h
```

palmer-dabbelt, 2017-08-21:
I had that floating around my tree somewhere, but I also think it seems
wrong.  We'll just see what happens next time I submit upstream, I'm sure
they'll know... 😎

closed 2017-08-21 by palmer-dabbelt

kito-cheng, 2017-08-22:
OK, so I just add work around in my local repo :P