riscvarchive/riscv-linux #83: Failed to boot linux kernel

bensenq, 2017-08-28:
Does spike of the  current version  support boot linux kernel?

I've built bbl with linux kernel as its payload, and then run `spike bbl` , but no kernel or user message show up, why? 

Branches:

- riscv-fesrv: master
- riscv-isa-sim: master
- riscv-poky: master
- riscv-linux: riscv-next

bensenq, 2017-08-28:
By the way,  `spike pk hello` gives right message.

bensenq, 2017-10-02:
The spike is in the infinite loop:

```
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           badaddr 0x0000000000000000
```

I think it's a version mismatch.

palmer-dabbelt, 2017-10-04:
There should be matching versions in github.com/sifive/freedom-u-sdk on the "new" branch, which we'll merge when we have a handful of issues sorted out.  I'm going to close this.

closed 2017-10-04 by palmer-dabbelt