riscvarchive/riscv-linux #85: Unable to boot vmlinux with spike

szhou42, 2017-09-01:
Hi I have built the 32-bit riscv toolchain and riscv-linux using the following commands
cd riscv-tools
./build-rv32ima.sh

cd riscv-gnu-toolchain/build
../configure --prefix=$RISCV --with-arch=rv32gc --with-abi=ilp32d
make linux

cd ~/riscv-linux
make ARCH=riscv defconfig
make -j16 ARCH=riscv CROSS_COMPILE=riscv32-unknown-linux-gnu-

However, when I do "spike bbl vmlinux", spike hangs.
spike -l bbl vmlinux shows me this exception:
"core   0: exception trap_instruction_access_fault, epc 0x0000000000000000"

Also, compiling and running a hello world program with the riscv32-unknown-linux-gnu-gcc also make spike hangs but does not produce such exception.

What could be the cause of this?

palmer-dabbelt, 2017-09-01:
This is probably a version mismatch.  The versions in here should match

  https://github.com/sifive/freedom-u-sdk/tree/new

If you just run "make sim" it should boot Linux.

szhou42, 2017-09-02:
Hi palmer, i don't see the target "sim" in any makefile, so i compiled each manually.

cd riscv-gnu-toolchain
./configure --prefix=$RISCV --with-arch=rv32gc --with-abi=ilp32d
make linux

this give me the following error message:
https://pastebin.com/7yJ7f3sd

It seems to ignore my configuration to build a 32-bit compiler and build a 64-bit one instead.

szhou42, 2017-09-03:
I have built the riscv32-unknown-linux-gnu toolchain, and trying to build linux with this command

make -j4 ARCH=riscv vmlinux CROSS_COMPILE=riscv32-unknown-
 
and got lots of incompatible ABI errors for these .o files

/home/osdev/install/riscv/lib/gcc/riscv32-unknown-linux-gnu/7.1.1/../../../../riscv32-unknown-linux-gnu/bin/ld: arch/riscv/kernel/vdso/rt_sigreturn.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/osdev/install/riscv/lib/gcc/riscv32-unknown-linux-gnu/7.1.1/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file arch/riscv/kernel/vdso/rt_sigreturn.o
/home/osdev/install/riscv/lib/gcc/riscv32-unknown-linux-gnu/7.1.1/../../../../riscv32-unknown-linux-gnu/bin/ld: arch/riscv/kernel/vdso/rt_sigreturn.o: file class ELFCLASS64 incompatible with ELFCLASS32
/home/osdev/install/riscv/lib/gcc/riscv32-unknown-linux-gnu/7.1.1/../../../../riscv32-unknown-linux-gnu/bin/ld: final link failed: File in wrong format
collect2: error: ld returned 1 exit status
  VDSOLD  arch/riscv/kernel/vdso/vdso-dummy.o

palmer-dabbelt, 2017-11-16:
Everything should line up now in freedom-u-sdk, and we recently added support for setting the ISA and ABI.  YMMV on the 32-bit port actually working, though, as it doesn't get tested as much as it should :).

closed 2017-11-16 by palmer-dabbelt