riscvarchive/riscv-linux #87: Port Linux to non-virtual memory CPU

jwang-us, 2017-09-27:
Hi,

If we don't support virtual memory in RISC-V, can we still use riscv-linux? Is there any switch while compiling Linux kernel to support non-virtual memory CPU?

Thank you.

Jim

palmer-dabbelt, 2017-09-27:
We don't currently have a no-MMU Linux port for RISC-V.  While it would be possible to build one, it's a big chunk of work and to the best of my knowledge nobody is working on it.  If you're interested in working on the port then I can try to help out, but otherwise I don't see it happening any time soon.

haizzus, 2018-06-01:
@palmer-dabbelt I'm interested and considering spending a few hours a week on this. How do I contact you (email/slack ?) to discuss further?

palmer-dabbelt, 2018-06-07:
#riscv on freednode or linux-riscv@lists.infradead.org are the best bets -- Linux is open source and there's people who aren't just me working on it, so there you'll find everyone.