riscvarchive/riscv-linux #94: Running "spike bbl" command freezing  in the current build

vinod1993, 2017-10-29:
Hello,
       I built the complete Linux environment by following the steps provided in this as well as the ones given in riscv-tools repository. I did the following steps.
       1. Built the busybox utility with all the suggested menu-configurations. There were two ways of getting the initramfs.txt and inittab files (one in riscv-tools and the other here). I used the steps from this repository. 
        2. Successfully built the linux kernel image (vmlinux) and bbl (after recompiling both riscv-linux and riscv-pk).
However, When I run the command spike bbl, it just freezes and when you try to come out, Spike goes to debug mode instead of printing stuffs on the screen. There also seems to be a trap instruction access fault. 
        3. I have another build where I followed the same steps, on a previous version of the riscv-tools  which seems to work fine, where the bootup process goes till the ash terminal invocation. Please help.

palmer-dabbelt, 2017-10-30:
This is usually a version mismatch between Spike and Linux.  Can you try the versions tagged on the "new" branch of github.com/sifive/freedom-u-sdk?

vinod1993, 2017-10-30:
Thanks for the reply Palmer. I will try and let you know.

palmer-dabbelt, 2017-11-16:
There's a coherent set of versions tagged in freedom-u-sdk, if that doesn't work then feel free to re-open the bug.

closed 2017-11-16 by palmer-dabbelt

edcote, 2017-11-30:
this is a failure in my CI system, any tips on how one would root cause the issue? or, at the very least, get a trace, error msg, that I can use for triage?

palmer-dabbelt, 2017-11-30:
Well, you've got the illegal instruction right there in epc.  You should be able to figure out what function it's in by looking at your Linux binary, if you can upload an "riscv*-objdump -d vmlinux" then I can take a look.