riscvarchive/riscv-linux #120: Build fails with lack of auto-generated headers

jleidel, 2017-12-28:
- RISCV-Tools master from 12/27
- RISCV-Linux master from 12/27
- The only modification to the kernel config was to bump to max CPUs to 32 (from 8)
- Built kernel headers from latest source above
- Built riscv-unknown-linux-gnu-* toolchain after headers were installed
- Built kernel using: 
$> make -j16 ARCH=riscv CROSS_COMPILE=riscv-unknown-linux-gnu-

The build errors out with the following (I also verified that the bug exists with '-j1'): 

HOSTCC  scripts/genksyms/lex.lex.o
In file included from ./arch/riscv/include/asm/smp.h:19:0,
                 from ./include/linux/smp.h:64,
                 from ./include/linux/percpu.h:7,
                 from ./include/linux/percpu-rwsem.h:7,
                 from ./include/linux/fs.h:33,
                 from /hdd/home/jleidel/dev/working/riscv-linux/scripts/gdb/linux/constants.py.in:15:
./arch/riscv/include/asm/asm-offsets.h:1:10: fatal error: generated/asm-offsets.h: No such file or directory
 #include <generated/asm-offsets.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scripts/gdb/linux/Makefile:19: recipe for target 'scripts/gdb/linux/constants.py' failed
make[2]: *** [scripts/gdb/linux/constants.py] Error 1
Kbuild:81: recipe for target 'build_constants_py' failed
make[1]: *** [build_constants_py] Error 2
Makefile:1057: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
make: *** Waiting for unfinished jobs....

palmer-dabbelt, 2017-12-28:
`riscv-unknown-linux-gnu-` isn't usually what we call the compilers, it's usually something more like `riscv64-unknown-linux-gnu-`.

jleidel, 2017-12-28:
Palmer, sorry... that was a typo on my part.  My cross compilation prefix is currently set to: riscv64-unknown-linux-gnu-

palmer-dabbelt, 2017-12-28:
I don't know, then.  asm-offsets.h is a particularly special file (it's generated), so it's possible something is a bit wrong in our build system.  Does a `make mrproper` fix this?  I've seen some oddness when bouncing between build directories or ARCH values before.

jleidel, 2017-12-28:
I just pulled a fresh tree from master and tried rebuilding (with make mrproper to force a clean), no luck.  

$> echo $CCPREFIX
riscv64-unknown-linux-gnu-

$> make ARCH=riscv CROSS_COMPILE=$CCPREFIX
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/riscv/include/generated/uapi/asm/setup.h
  WRAP    arch/riscv/include/generated/uapi/asm/unistd.h
  WRAP    arch/riscv/include/generated/uapi/asm/bpf_perf_event.h
  WRAP    arch/riscv/include/generated/uapi/asm/errno.h
  WRAP    arch/riscv/include/generated/uapi/asm/fcntl.h
  WRAP    arch/riscv/include/generated/uapi/asm/ioctl.h
  WRAP    arch/riscv/include/generated/uapi/asm/ioctls.h
  WRAP    arch/riscv/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/riscv/include/generated/uapi/asm/mman.h
  WRAP    arch/riscv/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/riscv/include/generated/uapi/asm/param.h
  WRAP    arch/riscv/include/generated/uapi/asm/poll.h
  WRAP    arch/riscv/include/generated/uapi/asm/posix_types.h
  WRAP    arch/riscv/include/generated/uapi/asm/resource.h
  WRAP    arch/riscv/include/generated/uapi/asm/sembuf.h
  WRAP    arch/riscv/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/riscv/include/generated/uapi/asm/signal.h
  WRAP    arch/riscv/include/generated/uapi/asm/socket.h
  WRAP    arch/riscv/include/generated/uapi/asm/sockios.h
  WRAP    arch/riscv/include/generated/uapi/asm/stat.h
  WRAP    arch/riscv/include/generated/uapi/asm/statfs.h
  WRAP    arch/riscv/include/generated/uapi/asm/swab.h
  WRAP    arch/riscv/include/generated/uapi/asm/termbits.h
  WRAP    arch/riscv/include/generated/uapi/asm/termios.h
  WRAP    arch/riscv/include/generated/uapi/asm/types.h
  WRAP    arch/riscv/include/generated/asm/bugs.h
  WRAP    arch/riscv/include/generated/asm/cacheflush.h
  WRAP    arch/riscv/include/generated/asm/checksum.h
  WRAP    arch/riscv/include/generated/asm/clkdev.h
  WRAP    arch/riscv/include/generated/asm/cputime.h
  WRAP    arch/riscv/include/generated/asm/device.h
  WRAP    arch/riscv/include/generated/asm/div64.h
  WRAP    arch/riscv/include/generated/asm/dma.h
  WRAP    arch/riscv/include/generated/asm/dma-contiguous.h
  WRAP    arch/riscv/include/generated/asm/emergency-restart.h
  WRAP    arch/riscv/include/generated/asm/errno.h
  WRAP    arch/riscv/include/generated/asm/exec.h
  WRAP    arch/riscv/include/generated/asm/fb.h
  WRAP    arch/riscv/include/generated/asm/fcntl.h
  WRAP    arch/riscv/include/generated/asm/ftrace.h
  WRAP    arch/riscv/include/generated/asm/futex.h
  WRAP    arch/riscv/include/generated/asm/hardirq.h
  WRAP    arch/riscv/include/generated/asm/hash.h
  WRAP    arch/riscv/include/generated/asm/hw_irq.h
  WRAP    arch/riscv/include/generated/asm/ioctl.h
  WRAP    arch/riscv/include/generated/asm/ioctls.h
  WRAP    arch/riscv/include/generated/asm/ipcbuf.h
  WRAP    arch/riscv/include/generated/asm/irq_regs.h
  WRAP    arch/riscv/include/generated/asm/irq_work.h
  WRAP    arch/riscv/include/generated/asm/kdebug.h
  WRAP    arch/riscv/include/generated/asm/kmap_types.h
  WRAP    arch/riscv/include/generated/asm/kvm_para.h
  WRAP    arch/riscv/include/generated/asm/local.h
  WRAP    arch/riscv/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/riscv/include/generated/asm/mman.h
  WRAP    arch/riscv/include/generated/asm/module.h
  WRAP    arch/riscv/include/generated/asm/msgbuf.h
  WRAP    arch/riscv/include/generated/asm/mutex.h
  WRAP    arch/riscv/include/generated/asm/param.h
  WRAP    arch/riscv/include/generated/asm/percpu.h
  WRAP    arch/riscv/include/generated/asm/poll.h
[... 28 lines cut ...]
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  CHK     include/generated/timeconst.h
  UPD     include/generated/timeconst.h
  CC      kernel/bounds.s
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
  GEN     scripts/gdb/linux/constants.py
In file included from ./arch/riscv/include/asm/smp.h:19:0,
                 from ./include/linux/smp.h:64,
                 from ./include/linux/percpu.h:7,
                 from ./include/linux/percpu-rwsem.h:7,
                 from ./include/linux/fs.h:33,
                 from /hdd/home/jleidel/dev/working/riscv-linux/scripts/gdb/linux/constants.py.in:15:
./arch/riscv/include/asm/asm-offsets.h:1:10: fatal error: generated/asm-offsets.h: No such file or directory
 #include <generated/asm-offsets.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scripts/gdb/linux/Makefile:19: recipe for target 'scripts/gdb/linux/constants.py' failed
make[2]: *** [scripts/gdb/linux/constants.py] Error 1
Kbuild:81: recipe for target 'build_constants_py' failed
make[1]: *** [build_constants_py] Error 2
Makefile:1057: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2

jleidel, 2017-12-28:
I found the correct recipe of commands.  Posting here for archival purposes: 
$> export RISCV=/path/to/riscv/install
$> export PATH=$PATH:$RISCV/bin
$> export CCPREFIX=riscv64-unknown-linux-gnu-
$> make mrproper
$> make ARCH=riscv CROSS_COMPILE=$CCPREFIX defconfig
$> make ARCH=riscv CROSS_COMPILE=$CCPREFIX menuconfig
// Enable SMP build and set max cores to 32
$> make ARCH=riscv CROSS_COMPILE=$CCPREFIX

closed 2017-12-28 by jleidel

lineprinter, 2018-09-04:
Since the build fails because GDB scripts cannot be built (because the build system attempts to create them too soon, it seems), `CONFIG_GDB_SCRIPTS=n` fixes it for me.