riscvarchive/riscv-linux #121: what branch or tag is suitable for riscv-qemu

EvaChenn, 2018-01-05:
Hi, I link to here due to riscv-qemu
in the riscv-qemu's README, they used **priv-1.9 branch of riscv-linux** as an example for building linux image for riscv-qemu
However, there is no priv-1.9 branch of riscv-linux.
I have tried branch riscv-linux-4-14 of riscv-linux, but I got hanging after doing this command
_$  ~riscv-qemu/riscv64-softmmu/qemu-system-riscv64 -kernel ~/riscv-tools/riscv-pk/build/bbl -m 1024M -nographic_

is there any tutorial for building a riscv-linux for riscv-qemu? or any suggestion?
thanks!!

palmer-dabbelt, 2018-01-06:
I just merged QEMU support into freedom-u-sdk's master branch.  It's still a bit rough around the edges, but on a clean build I can "make qemu" and get into a shell with a DHCP lease.

michaeljclark, 2018-02-25:
You can use the riscv-qemu branch in freedom-u-sdk. If you do, make sure to unset RISCV environment variable if you have it set (as it needs a specific version of riscv-gnu-toolchain), then run `make qemu` and the SDK will build qemu, a buildroot rootfs and linux-kernel, then start it up:

```
unset RISCV
git clone --recursive https://github.com/sifive/freedom-u-sdk.git
cd freedom-u-sdk
make qemu
```

If you want SMP support, you'll need a more recent version of QEMU. The default riscv-qemu `riscv-next` branch has relatively solid SMP support:

- https://github.com/riscv/riscv-qemu/tree/riscv-next
- https://github.com/sifive/freedom-u-sdk/pull/35

To build an SMP linux-kernel, run menuconfig, enable "Platform Type" -> "Symmetric Multi-Processing"

```
make ARCH=riscv menuconfig
```

To enable smp in QEMU, add `-smp cpus=4` to the command line.

I've tested SMP kernels in latest qemu using the `riscv-linux-4.14` and `riscv-linux-4.15` branches.

You can also take a look at the QEMU wiki. I'll add `freedom-u-sdk` instructions to the QEMU wiki, no i've figured out how to get it building (you must unset RISCV environment variable).

- https://github.com/riscv/riscv-qemu/wiki