riscvarchive/riscv-linux #130: WARNING: CPU: 0 PID: 0 at drivers/irqchip/irq-riscv-intc.c:158 irq_enable+0x32/0x58

michaeljclark, 2018-02-25:
Kernel warning when booting `riscv-linux-4.15` branch in latest QEMU with SMP enabled.

```
static void riscv_irq_enable(struct irq_data *d)
{
        struct riscv_irq_data *data = irq_data_get_irq_chip_data(d);

        /*
         * It's only possible to write SIE on the current hart.  This jumps
         * over to the target hart if it's not the current one.  It's invalid
         * to write SIE on a hart that's not currently running.
         */
        if (data->hart == smp_processor_id())
                riscv_irq_unmask(d);
        else if (cpu_online(data->hart))
                riscv_remote_ctrl(data->hart, riscv_irq_enable_helper, d);
        else
                WARN_ON_ONCE(1); // <- this warning is triggering 
}
```

It warns only once, so we don't know how often it is triggering. The kernel appears stable.

```
$ sudo ./riscv64-softmmu/qemu-system-riscv64 -smp cpus=4 -nographic -machine virt -kernel qemu-images/bbl-4.15 -append "root=/dev/vda ro" -drive file=qemu-images/busybear.bin,format=raw,id=hd0 -device virtio-blk-device,drive=hd0 -netdev type=tap,script=./qemu-images/ifup.macos,downscript=./qemu-images/ifdown.macos,id=net0 -device virtio-net-device,netdev=net0
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00050-g55f79e25d599 (mclark@minty) (gcc version 7.1.1 20170509 (GCC)) #96 SMP Mon Feb 26 09:20:44 NZDT 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000]   Normal   [mem 0x0000000088000000-0x0000087fffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] software IO TLB [mem 0x83e3b000-0x87e3b000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s24600 r0 d32744 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 31815
[    0.000000] Kernel command line: root=/dev/vda ro
[    0.000000] Dentry cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 57404K/129024K available (2245K kernel code, 163K rwdata, 563K rodata, 124K init, 266K bss, 71620K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] ------------[ cut here ]------------
[    0.000000] WARNING: CPU: 0 PID: 0 at drivers/irqchip/irq-riscv-intc.c:158 irq_enable+0x32/0x58
[    0.000000] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.15.0-00050-g55f79e25d599 #96
[    0.000000] Call Trace:
[    0.000000] [<        (ptrval)>] walk_stackframe+0x0/0x7e
[    0.000000] [<        (ptrval)>] dump_stack+0x5a/0x76
[    0.000000] [<        (ptrval)>] __warn+0xc0/0xda
[    0.000000] [<        (ptrval)>] irq_enable+0x2e/0x58
[    0.000000] [<        (ptrval)>] __irq_startup+0x82/0x8a
[    0.000000] [<        (ptrval)>] irq_startup+0x4c/0xf4
[    0.000000] [<        (ptrval)>] plic_irq_disable+0x6e/0x72
[    0.000000] [<        (ptrval)>] irq_set_chained_handler_and_data+0x24/0x42
[    0.000000] [<        (ptrval)>] plic_init+0x1e6/0x312
[    0.000000] [<        (ptrval)>] printk+0x1e/0x28
[    0.000000] [<        (ptrval)>] riscv_intc_init+0xc6/0xf4
[    0.000000] [<        (ptrval)>] of_irq_init+0x112/0x206
[    0.000000] [<        (ptrval)>] start_kernel+0x22c/0x354
[    0.000000] random: get_random_bytes called from print_oops_end_marker+0x36/0x4e with crng_init=0
[    0.000000] ---[ end trace 811f0666af6c5eb2 ]---
[    0.000000] riscv,plic0,c000000: mapped 10 interrupts to 4/8 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.040000] Hierarchical SRCU implementation.
[    0.050000] smp: Bringing up secondary CPUs ...
[    0.050000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.070000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.070000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.080000] smp: Brought up 1 node, 4 CPUs
[    0.130000] devtmpfs: initialized
[    0.140000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.140000] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.150000] NET: Registered protocol family 16
[    0.170000] vgaarb: loaded
[    0.190000] clocksource: Switched to clocksource riscv_clocksource
[    0.200000] NET: Registered protocol family 2
[    0.200000] TCP established hash table entries: 1024 (order: 1, 8192 bytes)
[    0.200000] TCP bind hash table entries: 1024 (order: 2, 16384 bytes)
[    0.200000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.210000] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.220000] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.230000] NET: Registered protocol family 1
[    0.250000] workingset: timestamp_bits=62 max_order=14 bucket_order=0
[    0.300000] random: fast init done
[    0.370000] io scheduler noop registered
[    0.370000] io scheduler cfq registered (default)
[    0.380000] io scheduler mq-deadline registered
[    0.380000] io scheduler kyber registered
[    0.950000] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.980000] console [ttyS0] disabled
[    0.980000] 10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 13, base_baud = 230400) is a 16550A
[    0.980000] console [ttyS0] enabled
[    0.980000] console [ttyS0] enabled
[    0.990000] bootconsole [early0] disabled
[    0.990000] bootconsole [early0] disabled
[    1.190000] EXT4-fs (vda): couldn't mount as ext3 due to feature incompatibilities
[    1.200000] EXT4-fs (vda): couldn't mount as ext2 due to feature incompatibilities
[    1.220000] EXT4-fs (vda): INFO: recovery required on readonly filesystem
[    1.220000] EXT4-fs (vda): write access will be enabled during recovery
[    1.370000] EXT4-fs (vda): recovery complete
[    1.380000] EXT4-fs (vda): mounted filesystem with ordered data mode. Opts: (null)
[    1.390000] VFS: Mounted root (ext4 filesystem) readonly on device 254:0.
[    1.400000] devtmpfs: mounted
[    1.410000] Freeing unused kernel memory: 124K
[    1.410000] This architecture does not have kernel memory protection.
[    2.640000] EXT4-fs (vda): re-mounted. Opts: data=ordered
Initializing mdev...

ucbvax login: root
Password: ntpd: setting time to 2018-02-25 20:23:36.617367 (offset +1519590206.408055s)

    ____                   ____                     __    _                 
   / __ )__  _________  __/ __ )___  ____ ______   / /   (_)___  __  ___  __
  / __  / / / / ___/ / / / __  / _ \/ __ `/ ___/  / /   / / __ \/ / / / |/_/
 / /_/ / /_/ (__  ) /_/ / /_/ /  __/ /_/ / /     / /___/ / / / / /_/ />  <  
/_____/\__,_/____/\__, /_____/\___/\__,_/_/     /_____/_/_/ /_/\__,_/_/|_|  
                 /____/                                                     
root@ucbvax:~# 
root@ucbvax:~# 
root@ucbvax:~# cat /proc/cpuinfo 
hart	: 0
isa	: rv64imafdc
mmu	: sv48

hart	: 1
isa	: rv64imafdc
mmu	: sv48

hart	: 2
isa	: rv64imafdc
mmu	: sv48

hart	: 3
isa	: rv64imafdc
mmu	: sv48

root@ucbvax:~# 
```

sorear, 2018-02-26:
Fedora has seen the same warning.  We haven't investigated it since, as you say, the kernel appears stable.

michaeljclark, 2018-02-26:
I'm getting the same message on linux-kernel-4.14 running on the RTL on VC707 FPGA

```
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.14.0-00030-gc2d852c (hes@gamma04.internal.sifive.com) (gcc version 7.2.0 (GCC)) #1 SMP Tue Dec 19 16:13:56 P7
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0xffffffe00001e200 (5672960 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000] free_area_init_node: node 0, pgdat ffffffe0009425c0, node_mem_map ffffffe03f007000
[    0.000000]   DMA zone: 3577 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:31
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[    0.000000] elf_hwcap is 0x1105
[    0.000000] percpu: Embedded 14 pages/cpu @ffffffe03efbd000 s28376 r0 d28968 u57344
[    0.000000] pcpu-alloc: s28376 r0 d28968 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 1 [0] 2 [0] 3 [0] 4 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[    0.000000] Kernel command line: earlyprintk 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 1020056K/1046528K available (2826K kernel code, 236K rwdata, 746K rodata, 5695K init, 774K bss, 26472K reserved, 0K )
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=5, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU event tracing is enabled.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=5.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=5
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,4: 64 local interrupts mapped
[    0.000000] ------------[ cut here ]------------
[    0.000000] WARNING: CPU: 1 PID: 0 at /scratch/hes/freedom-u-sdk/linux/drivers/irqchip/irq-riscv-intc.c:157 riscv_irq_enable+0x92/0x94
[    0.000000] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.0-00030-gc2d852c #1
[    0.000000] Call Trace:
[    0.000000] [<ffffffe000591bfa>] walk_stackframe+0x0/0xa2
[    0.000000] [<ffffffe000591d38>] show_stack+0x26/0x34
[    0.000000] [<ffffffe00083e06e>] dump_stack+0x5e/0x7c
[    0.000000] [<ffffffe000594ef2>] __warn+0xbe/0xd4
[    0.000000] [<ffffffe000594f82>] warn_slowpath_null+0x10/0x1c
[    0.000000] [<ffffffe0006e6bde>] riscv_irq_enable+0x8e/0x94
[    0.000000] [<ffffffe0005c8104>] irq_enable+0x3a/0x6a
[    0.000000] [<ffffffe0005c817e>] __irq_startup+0x4a/0x52
[    0.000000] [<ffffffe0005c81d6>] irq_startup+0x50/0xda
[    0.000000] [<ffffffe0005c82d0>] __irq_do_set_handler+0x70/0x136
[    0.000000] [<ffffffe0005c8d0e>] irq_set_chained_handler_and_data+0x2c/0x4a
[    0.000000] [<ffffffe0006e6814>] plic_init+0x1f6/0x322
[    0.000000] [<ffffffe000011cd0>] of_irq_init+0x136/0x252
[    0.000000] [<ffffffe00000c7e6>] irqchip_init+0x10/0x1c
[    0.000000] [<ffffffe0000021dc>] init_IRQ+0x8/0x14
[    0.000000] [<ffffffe0000008e0>] start_kernel+0x20a/0x348
[    0.000000] [<ffffffe00000004e>] _sinittext+0x4e/0x56
[    0.000000] random: get_random_bytes called from print_oops_end_marker+0x3a/0x52 with crng_init=0
[    0.000000] ---[ end trace 242ad3eb94620356 ]---
[    0.000000] riscv,plic0,c000000: mapped 18 interrupts to 4/9 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.010000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[... 84 lines cut ...]
[    6.280000] e1000e 0000:01:00.0: enabling bus mastering
[    6.290000] e1000e 0000:01:00.0: Interrupt Throttling Rate (ints/sec) set to dynamic conservative mode
[    6.400000] e1000e 0000:01:00.0 0000:01:00.0 (uninitialized): registered PHC clock
[    6.530000] e1000e 0000:01:00.0 eth0: (PCI Express:2.5GT/s:Width x1) 00:1b:21:24:12:33
[    6.540000] e1000e 0000:01:00.0 eth0: Intel(R) PRO/1000 Network Connection
[    6.540000] e1000e 0000:01:00.0 eth0: MAC: 3, PHY: 8, PBA No: E46981-003
[    6.560000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    6.560000] ehci-pci: EHCI PCI platform driver
[    6.570000] usbcore: registered new interface driver usb-storage
[    6.580000] IR NEC protocol handler initialized
[    6.580000] IR RC5(x/sz) protocol handler initialized
[    6.590000] IR RC6 protocol handler initialized
[    6.590000] IR JVC protocol handler initialized
[    6.600000] IR Sony protocol handler initialized
[    6.600000] IR SANYO protocol handler initialized
[... cut ...] r initialized
[    6.610000] IR MCE Keyboard/mouse protocol handler initialized
[    6.620000] IR XMP protocol handler initialized
[    6.630000] usbcore: registered new interface driver usbhid
[    6.630000] usbhid: USB HID core driver
[    6.640000] NET: Registered protocol family 17
[    6.960000] Freeing unused kernel memory: 5692K
[    6.970000] This architecture does not have kernel memory protection.
[   34.850000] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
```

palmer-dabbelt, 2018-02-27:
This isn't a scary warning, it's just a bit of cleanup I need to do related to how we initialize our interrupt stuff.

atishp04, 2018-11-02:
This issue is no longer present after we got rid of local interrupt controller driver in 4.19. We can close this issue.