riscvarchive/riscv-linux #133: I can't do "make  ARCH=riscv".

yosukezith, 2018-04-25:
I can't do "make ARCH=riscv".
Please solve this problem anyone.

$ make  ARCH=riscv

  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CC      kernel/bounds.s
gcc: error: unrecognized argument in option '-mabi=lp64'
gcc: note: valid arguments to '-mabi=' are: ms sysv
gcc: error: unrecognized argument in option '-mcmodel=medany'
gcc: note: valid arguments to '-mcmodel=' are: 32 kernel large medium small
gcc: error: unrecognized command line option '-mno-save-restore'
Kbuild:21: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 1
Makefile:1112: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2

kito-cheng, 2018-04-25:
You need use CROSS_COMPILE to specify your toolchain path.
If your riscv compiler already in your path you can try:
`$ make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-`

yosukezith, 2018-04-25:
I can do it.
Thank you for solving this problem.

closed 2018-04-25 by yosukezith