riscvarchive/riscv-linux #134: Help

Royen102, 2018-04-27:


Royen102, 2018-04-27:
/home/david/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld：不支持 -shared
collect2: error: ld returned 1 exit status
  LD      arch/riscv/kernel/vdso/vdso-syms.o
  OBJCOPY arch/riscv/kernel/vdso/vdso.so
riscv64-unknown-elf-objcopy: arch/riscv/kernel/vdso/vdso.so.dbg：无此文件
arch/riscv/kernel/vdso/Makefile:48: recipe for target 'arch/riscv/kernel/vdso/vdso.so' failed
make[2]: *** [arch/riscv/kernel/vdso/vdso.so] Error 1
scripts/Makefile.build:559: recipe for target 'arch/riscv/kernel/vdso' failed
make[1]: *** [arch/riscv/kernel/vdso] Error 2
Makefile:1060: recipe for target 'arch/riscv/kernel' failed
make: *** [arch/riscv/kernel] Error 2
make: *** 正在等待未完成的任务....
  CC      init/initramfs.o
  CC      init/calibrate.o
  CC      init/init_task.o
  CC      init/version.o
  AR      init/built-in.a

kito-cheng, 2018-04-27:
It seem like because -shared and -pie support is removed from riscv elf binutils. So that's mean you must use linux toolchain to compile linux kernel.

@palmer-dabbelt @jim-wilson just curious why those option are removed in elf bare-metal toolchain?

Royen102, 2018-04-27:
@kito-cheng  Thank U very much！As you say, solve the problem successfully.

closed 2018-04-27 by Royen102

jim-wilson, 2018-04-27:
RISC-V newlib does not have -shared support.  This was causing binutils testsuite failures, as the riscv*-elf toolchain claimed to have shared support, but it did not work.  Disabling shared support solved the problem.  Most bare metal toolchains have shared support turned off, so this is normal behavior.

Note that linux and bare metal toolchains often have different ABIs.  Using an elf toolchain on a linux kernel is risky.  You really should use the linux toolchain to ensure that you get the right ABI.