riscvarchive/riscv-linux #138: patch for PTRACE_SETREGSET bug

jim-wilson, 2018-06-10:
This was found while trying to get a native gdb port working.  There is a trivial typo in riscv_gpr_set.  Compare with the code in riscv_gpr_get.

```
diff --git a/arch/riscv/kernel/ptrace.c b/arch/riscv/kernel/ptrace.c
index ba3e80712797..9f82a7e34c64 100644
--- a/arch/riscv/kernel/ptrace.c
+++ b/arch/riscv/kernel/ptrace.c
@@ -50,7 +50,7 @@ static int riscv_gpr_set(struct task_struct *target,
        struct pt_regs *regs;
 
        regs = task_pt_regs(target);
-       ret = user_regset_copyin(&pos, &count, &kbuf, &ubuf, &regs, 0, -1);
+       ret = user_regset_copyin(&pos, &count, &kbuf, &ubuf, regs, 0, -1);
        return ret;
 }
 
```

palmer-dabbelt, 2018-06-11:
Thanks.  The kernel needs `Signed-off-by`, do you mind submitting a patch with one?

jim-wilson, 2018-06-11:
I don't know anything about submitting linux kernel patches.  Do you want me to email one somewhere?  Submit a pull request here?  Or something else?

jim-wilson, 2018-06-11:
I found some instructions on how to submit a kernel patch and tried following them, which means my patch was sent via email to about 6 addresses.

palmer-dabbelt, 2018-06-11:
I got it, thanks.  I'm going to close this here.

closed 2018-06-11 by palmer-dabbelt