riscvarchive/riscv-linux #140: icache not flushed correctly

zeldin, 2018-07-11:
It appears that the flush_icache VDSO is not working correctly on Freedom U540.
The following test program:
```C
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <string.h>
#include <sys/mman.h>
 
static void *allocate_code(void)
{
  void *addr = aligned_alloc(4096, 4096);
  mprotect(addr, 4096, PROT_EXEC | PROT_READ | PROT_WRITE);
  memset(addr, 0, 4096);
  return addr;
}
 
static void generate_code(void *p, int n)
{
  uint16_t *code = p;
 
  /* Assumes 0 <= n <= 31 */
  code[0] = 0x4501 | (n<<2); /* c.li a0, n */
  code[1] = 0x8082; /* c.jr ra */
 
  __builtin___clear_cache(code, code+16);
}
 
static int call_code(void *p)
{
  return (*(int (*)(void))p)();
}
 
int main()
{
  void *addr = allocate_code();
 
  int a, b;
 
  generate_code(addr, 3);
  a = call_code(addr);
 
  generate_code(addr, 7);
  b = call_code(addr);
 
  printf("Results: %d %d\n", a, b);
 
  return 0;
}
``` 
prints "3 3" when run on the HiFive Unleashed, but should print "3 7".

I'm building the kernel from the riscv-linux-4.15 branch.

zeldin, 2018-07-11:
Sure. 

> The generated ELF, to make sure it calls into glibc to perform the fence

It does.  Here is the generated code for the line that calls  `__builtin___clear_cache`:
```asm
/home/marcus/hack/hi5cache/main.c:23
   10588:       fe843783                ld      a5,-24(s0)
   1058c:       02078713                addi    a4,a5,32
   10590:       fe843783                ld      a5,-24(s0)
   10594:       4601                    li      a2,0
   10596:       85ba                    mv      a1,a4
   10598:       853e                    mv      a0,a5
   1059a:       7ce140ef                jal     ra,24d68 <__riscv_flush_icache>
```
> Checking the generated glibc to see if __riscv_flush_icache actually calls into the vDSO.

Yes.  There is first some code to check if something called `cached_func.7305` is non-NULL (I assume this is generated by the INLINE_SYSCALL macro), but if it is NULL the code continues and further down there is indeed a call to `_dl_vdso_vsym`.  I also checked that the binary does contain the string `__vdso_flush_icache`.

Now I had a clever idea though:  I ran strace on the binary.  This was the result:
```
execve("./main", ["./main"], 0x3fffacf030 /* 46 vars */) = 0
brk(NULL)                               = 0x75000
brk(0x75f70)                            = 0x75f70
uname({sysname="Linux", nodename="vanille", ...}) = 0
readlinkat(AT_FDCWD, "/proc/self/exe", "/home/marcus/hack/hi5cache/main", 4096) = 31
brk(0x96f70)                            = 0x96f70
brk(0x97000)                            = 0x97000
mprotect(0x77000, 4096, PROT_READ|PROT_WRITE|PROT_EXEC) = 0
riscv_flush_icache(0x77000, 0x77020, 0) = -1 ENOSYS (Function not implemented)
riscv_flush_icache(0x77000, 0x77020, 0) = -1 ENOSYS (Function not implemented)
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
write(1, "Results: 3 7\n", 13Results: 3 7
)          = 13
exit_group(0)                           = ?
+++ exited with 0 +++
```
Huh?  Function not implemented?  That doesn't sound right...  (The extra interferrence of ptrace actually made the program output the correct result, hilariously enough :-)

> Checking to see if the kernel's vDSO actually calls into sys_riscv_flush_icache at some point. The code in Linux to handle this lives in sys_riscv.c and looks like

Yup, looks exactly the same for me.  I'm at 2751b6aad, which is almost the tip of the riscv-linux-4.15 branch.

> Checking to see that Linux correctly performs the remote fences. There's some bookkeeping in here to avoid fences on harts that don't need them, and that could be broken. This code lives in smp.c and looks like

Well, my test program is single-threaded so that should not be an issue.  The ENOSYS seems like a much more likely candidate.  :-)

zeldin, 2018-07-11:
I did another test program which I compiled with `-Wl,-wrap,_dl_vdso_vsym`, in order to check that `_dl_vdso_vsym` is actually called (man, a working native gdb would be swell... ;-):
```C
#include <stdio.h>

void *__wrap__dl_vdso_vsym(const char *a, void *b)
{
  void *p;
  void *(*real_dl_vdso_vsym) (const char *name, void *version) =
    (void*)0x1057e;

  printf(">>VDSO_VSYM \"%s\"\n", a);
  p = real_dl_vdso_vsym(a, b);
  printf("<<VDSO_VSYM \"%s\" %p\n", a, p);
  return p;
}

int main()
{
  __builtin___clear_cache(main, main);

  return 0;
}
```
The address `0x1057e` was taking from an nm of the resulting binary (the actual symbol is hidden..).
The result of running this program was:
```
>>VDSO_VSYM "__vdso_flush_icache"
<<VDSO_VSYM "__vdso_flush_icache" 0x200000083c
```
So yes, `_dl_vdso_vsym("__vdso_flush_icache")` is called, and it does return non-NULL.  Checking with strace reveals that the failing syscall is still invoked.

zeldin, 2018-07-11:
I checked the disassembly of `__riscv_flush_icache` a little closer, and what it does is this:
```
40(sp) = __stack_chk_guard
a5 = cached_func.7305
if a5 != NULL {
  label1:
  (*a5)(arg0, arg1)
  if 40(sp) != __stack_chk_guard
     __stack_chk_fail()
  return
}
var.hash = 182943605
var.hidden = 1
var.name = "LINUX_4.15"
if var.hash != _dl_elf_hash (var.name)
  __assert_fail("linux_version.hash == _dl_elf_hash (linux_version.name)", "../sysdeps/unix/sysv/linux/riscv/flush-icache.c", 29, "__lookup_riscv_flush_icache")
var.filename = NULL
a5 = _dl_vdso_vsym("__vdso_flush_icache", &var)
if a5 != NULL {
 cached_func.7305 = a5
 goto label1
}
abort()
```
So unlike the code you pasted, there is no fallback to `__riscv_flush_icache_syscall`.  The function either calls the VDSO or `abort()`s.

So it all points to the actual VDSO being called, which in turn makes an unsupported system call?  I reran strace with `-i`, and sure enough:
```
[0000002000000844] riscv_flush_icache(0x10550, 0x10550, 0) = -1 ENOSYS (Function not implemented)
```
0x2000000844 is 8 bytes after the address returned by `_dl_vdso_vsym()`.

zeldin, 2018-07-11:
Hm, I think I see the problem.  In `arch/riscv/kernel/syscall_table.c`, this happens:
```C
#undef __SYSCALL
#define __SYSCALL(nr, call)     [nr] = (call),

void *sys_call_table[__NR_syscalls] = {
        [0 ... __NR_syscalls - 1] = sys_ni_syscall,
#include <asm/unistd.h>
};
```
`arch/riscv/include/asm/unistd.h` does indeed include `<uapi/asm/syscalls.h>`, which contains the definition of the syscall.  However, `arch/riscv/include/uapi/asm/syscalls.h` also contains an include guard `_ASM__UAPI__SYSCALLS_H`, and because the file has already been included once before, the syscall never gets added to the `sys_call_table`.  Removing the include guard should fix the issue.

zeldin, 2018-07-11:
I removed the include guard from `arch/riscv/include/uapi/asm/syscalls.h` and rebuilt the kernel.  Now the test program works correctly. :grinning:

aswaterman, 2018-07-12:
Nice. Palmer, can you incorporate this fix?

palmer-dabbelt, 2018-08-03:
https://lkml.org/lkml/2018/8/3/734

closed 2018-08-03 by palmer-dabbelt