riscvarchive/riscv-linux #150: What instruction sets do I need to run Linux on my own risc-v CPU?

hyf6661669, 2019-01-23:
I'm planning to design my own 32-bit risc-v cpu, put it into a FPGA and run Linux on it. What instruction sets does my cpu must support? Is RV32IMAC enough to boot Linux? Thank you!

aramoki, 2019-02-11:
I wonder that too,
I'm planning to design RISC64IMAC CPU I'm aware if it is capable to run Linux or not 
or what kind of other implementations need to make it possible?

davidlt, 2019-02-11:
IIRC the Linux kernel only requires I and A (check makefiles), but Linux distributions are targeting RV64GC as described in https://github.com/riscv/riscv-platform-specs/blob/master/riscv-unix.md

davidlt, 2019-02-11:
Note, that 32-bit Linux ABI is not yet frozen. Most likely that will happen with glibc release 2.30 (half a year).

aramoki, 2019-02-11:
so for a very basic implementation, RISC32IA or RISC64IA is enough to run Linux

aswaterman, 2019-02-11:
I’m not certain, but you might run into some minor issues if you omit the M
extension (missing emulation routines for mul/div). Should be pretty
straightforward to fix, if so.

palmer-dabbelt, 2019-02-11:
If you want to run the standard pre-compiled userspaces (ie, Debia/Fedora) then you'll need rv64gc.  The upstream Linux and glibc ports should work fine with rv64imac and rv64gc -- at least, that's what we test.  The 32-bit ports are coming along, but they won't be all the way there for another 6 months or so.

If you're willing to rebuild userspace but expect it to work, then your minimum is probably rv64ima.  There is no fundamental reason that rv64i won't boot Linux (on a single hart system), but that will require writing some code.

aramoki, 2019-02-12:
thanx for replies

closed 2019-02-15 by hyf6661669