riscvarchive/riscv-linux #151: Any plan for upstreaming DMA32?

dayeol, 2019-01-29:
Hi, @terpstra 

I'm currently trying to submit a patch for adding contiguous memory allocator (CMA) to RISC-V linux. I've already wrote some bit of code and tested it:

https://github.com/dayeol/riscv-linux/commit/16941d1d604abb33a43d7ced5836b4a5b374e295

I believe this should be implemented in `arch/riscv/mm/dma.c` in order to be migrated with architecture's dma allocators such as `dma_alloc_coherent`.

But I found that the file only exists in `riscv-linux-4.15` branch, and has not been upstreamed yet.
Do you have any plans to upstream `dma.c`?

Can you let me know how and where can I submit this patch, or if I'm missing something?

Thank you in advance!

terpstra, 2019-01-29:
Hi! I'm not the risc-v linux port maintainer.

@palmer-dabbelt @nvpw What do you think?

palmer-dabbelt, 2019-01-29:
You can submit the patch to linux-riscv@lists.infradead.org, based on Linus' latest master.  There aren't any concrete plans for anyone at SiFive to do this, but it is necessary to enable a large set of devices so it'll have to get done at some point.  If you're happy doing it that would be great, our development is all upstream now so that's the best way to get involved.