riscvarchive/riscv-linux #153: Are there any substantial differences between this repo and upstream?

marcopeereboom, 2019-04-11:
I have compiled this kernel and the latest stable upstream kernel and both appear to work. I was wondering if this repo has been or will be deprecated in the near future. Basically, I'd like to know which repo to fork and keep track of.

davidlt, 2019-04-11:
The development moved long time ago to "upstream locations".

Mailing list: http://lists.infradead.org/pipermail/linux-riscv/
Patchwork: https://patchwork.kernel.org/project/linux-riscv/list/

Upstream kernel is good enough to boot on QEMU, but will not work on, for example, SiFive Unleashed board due to missing drivers.

This is 5.1-rc1 with missing drivers: https://github.com/avpatel/linux/tree/v5.1-rc1_unleashed

Upstream tree location: https://git.kernel.org/pub/scm/linux/kernel/git/palmer/riscv-linux.git/
Palmer's tree: https://git.kernel.org/pub/scm/linux/kernel/git/palmer/linux.git/

Hope that helps.

marcopeereboom, 2019-04-11:
Yes it helps a lot! Looks like doco on the net is pretty much outdated everywhere.

closed 2019-04-11 by marcopeereboom